bigtreetech/BIGTREETECH-MINI-UPS-V2.0 #1: does not do anything

stefi01, 2019-11-04:
enabled with marlin 2.0 latest bug fix, wired to an skr 1.3 board, servo pin 2_0 but on power loss it provides a few seconds of power to move the motors but instead of moving the z axis up it just carry's on printing until it stops,  so it does not do anything but there is nothing in that code in config h or config adv to tell the z axis to move out of the way

stefi01, 2020-04-30:
after a few updates of marlin they finaly added the code in config adv to make this work and it did work for around december/januarys marlin bugfix updates, the latest update seems to work if it is not printing, as in i finished printing them turned off the printer and then the bed moves down but if the power is cut during a print then sometimes it works, sometimes not but in it does have to be all added within config.adv and switched on using the lcd.

TheMiguelBi, 2020-04-30:
I forgot the _M413 S1_ command !!
https://marlinfw.org/docs/gcode/M413.html

But now it seems to resume if it was printing with the **SD display**, while **onboard microSD** seems not to resume .....

TheMiguelBi, 2020-05-01:
> after a few updates of marlin they finaly added the code in config adv to make this work and it did work for around december/januarys marlin bugfix updates, the latest update seems to work if it is [...]

Out of curiosity, which display do you use ?? 🧐
I have done many other tests today. Here the detail: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/644#issue-610793577

stefi01, 2020-05-02:
> I have done many other tests today. Here the detail: bigtreetech/BIGTREETECH-TouchScreenFirmware#644
> (comment)

i use the bigtreetech tft with the wifi socket on the back, sumin .35 but
forgot but its pretty crap, flimsy and expensive for what it is