bigtreetech/BIGTREETECH-MINI-UPS-V2.0 #2: ups detect also power outage by starting the printer

TB1405, 2019-12-07:
I hvae a problem with the mini UPS 24V. 
The modul detect the power outage succesful and the Z Axis raise 10mm up. But if i power on the printer again the modul detect again a power outage and now the axis raises again 10mm up. But if i now continue the print the nozzle is 10mm above the obeject.

dziabi3, 2020-02-03:
I have the same problem. The printer responds when the power is turned on instead of a power outage during printing.
Did you deal with it?

TB1405, 2020-02-03:
I didn't bother with the UPS. Unfortunately I cannot help there

TheMiguelBi, 2020-04-30:
> Ho lo stesso problema. La stampante risponde quando viene accesa l'alimentazione anziché un'interruzione di corrente durante la stampa.
> L'hai affrontato?

Have you found any solution?
Even my print resumes with a difference of 10mm ....

TheMiguelBi, 2020-05-01:
Out of curiosity, which display do you use ?? 🧐
I have done many other tests today. 
Here the detail: bigtreetech/BIGTREETECH-TouchScreenFirmware#644

atharvbadkas, 2020-12-18:
Hi...I contacted BTT And they sent me files claiming they have tested it on BTT SKR MINI E3V2 AND BTT UPS 24V for Ender 3..but I'm still facing same issue with their firmware also....can u test it on your machine...and let me know??.... thanks...link https://drive.google.com/folderview?id=1lLJ8wB38pCyJNlhxEXdSRXJNUCHG2Eis