bigtreetech/BIGTREETECH-MINI-UPS-V2.0 #3: How to power off

CristianoGorla, 2020-01-23:
I don't know if it is an issue, but now, when I want to poweroff the printer, I must wait for complete discharge of the capacitors and this doesn't allow an emergency stop. Am I missing something or this is the wanted behavior

katherinehackworth, 2020-04-07:
Seconded. Even when not printing the UPS appears to activate and the printer takes 30-60 seconds to shut down. Appears the mini UPS is doing its job even when there _isn't_ a print going :/ Is this intended or something to be fixed

neuzele, 2020-04-28:
Im having the same issue, can somebody please give an answer?

Im close to throw it away or re-use the components

TheMiguelBi, 2020-04-30:
I'm also stuck here.
Inserting a switch would cause the capacitors to remain charged.... 🤔

LyWill, 2020-07-16:
Also having the same issue here. Pondering a solution and let me know if any of you have tried this. 
Since I also have the BTT Relay v1.2, I will always have 5V power to my SKRv1.3. I think what I am going to try to do is put a relay inline with 24V+ going to the SKR and wire the signal pin on the relay with the signal pin on the BTT Relay. So when I issue a M81 gcode command it shuts down the mains power as well as switch off the relay to 24V. This should 'turn off' the printer. The remaining power in the CAPS will be depleted by the PSU fan, which from what I can see also uses the 24V power line.
My issue was the CAPS would keep the SKR board powered for a full 60secs or more when I needed to reboot the board for firmware updates. When there are a few cycles of config changes and reflashing the board, this gets really old to wait for the board to power off. At times I would have to unplug the UPS to force the board to shut down immediately.

KL-XL, 2021-01-25:
Ditto

Ejay-3D, 2021-02-09:
Same issue here, Btt should have think about this , having to wait when shutting down the machine on purpose is a shame. ... I love btt products but most of them are half finished ... One more I guess.

Stefkeys, 2021-06-19:
I think the only option is to unplug the pwr-det pin (or wherever you have it connected and configured). The module isn't smart enough to know if the printer has been turned out by flipping the switch, power outage, m81, or otherwise. It only knows if power has been lost from the Mainboard or PSU

timknowlden, 2023-05-02:
Installed both of these modules this weekend, it is infurating not being able to perform a quick shutdown, especially if there is a problem with the printer.