bigtreetech/BIGTREETECH-MINI-UPS-V2.0 #9: How long to normally discharge UPS after Ender 3 Pro power off?

Ve6hgw, 2020-08-19:
SKR Mini E3 V2.0 and TFT35 E3 V3.00.  When I power off my printer with the on/off switch the fan keeps running for about 45 seconds then the TFT35 stays on for about another 15 seconds before it turns off.  Is this normal operation for the 24 volt Mini UPS?  Is this potentially damaging to any electrical components?

sergd1970, 2020-09-15:
Is this really the correct operation of mini UPS, that after turning off the printer does not turn off immediately, but works until the capacitors are exhausted?

KL-XL, 2021-01-25:
Ditto

timknowlden, 2023-05-02:
Any solution, it seems like a design flaw if command to switch printer off is used, id expect instant shut down and only when power tripped the capacitors should drain? When combined with BTT Relay module I cannot get the power off quick.