bigtreetech/BIGTREETECH-MINI-UPS-V2.0 #10: Btt ups doesn't work on versions above 2.0.5

berrenger, 2020-10-27:
Btt ups doesn't work on versions above 2.0.5.
I was able to compile and run it with Marlin version 2.0.5, but it doesn't work stable.  I could not manage to run it on higher versions.

berrenger, 2020-10-27:
My hardware: skr 1.4 turbo, btt tmc 2209, tft 35 v3, bltouch 3.1

HTTYD9, 2020-11-12:
I have same problem ,
in tft mode UPS is not working (power loss recovery).
[only relay works good , but relay with UPS not working]

BTT
skr pro v1.1
tft35 v3.0
UPS 24V v1.0
relay v1.2

berrenger, 2020-11-12:
Use marlin 2.0.5

ilkersismanoglu, 2020-11-19:
somehow i managed to get it worked.

hardware: skr 1.4 t, relay and mini ups 24V, tft35 v3.0
software: marlin bugfix 2.0.7

all marlin configs are done according to instructions, i tried several combinations but whenever i cut the power, z didn't raise.

finally switched to marlin emulator on my tft35, in config menu i've turned on power outage and save, started a print over sd, cut the power a few minutes later and voila!

in conf_advc.

`#define POWER_LOSS_RECOVERY
  #if ENABLED(POWER_LOSS_RECOVERY)
    #define PLR_ENABLED_DEFAULT   false 
    #define BACKUP_POWER_SUPPLY       
    //#define POWER_LOSS_RECOVER_ZHOME  
    #define POWER_LOSS_ZRAISE       10 
    #define POWER_LOSS_PIN         P1_25 // i used e1det port instead of pwrdet port
    #define POWER_LOSS_STATE     HIGH 
    #define POWER_LOSS_PULL           
    #define POWER_LOSS_PURGE_LEN   20 
    #define POWER_LOSS_RETRACT_LEN 10`

berrenger, 2020-11-19:
> bir şekilde onu çalıştırmayı başardım.
> 
> donanım: skr 1.4 t, röle ve mini ups 24V, tft35 v3.0 
> yazılımı: marlin bugfix 2.0.7
> 
> tüm marlin yapılandırmaları talimatlara göre yapılıyor, birkaç kombinasyon denedim ama gücü ne zaman kessem, z yükseltmedi.
> 
> nihayet tft35'imde marlin emülatörüne geçtim, yapılandırma menüsünde elektrik kesintisini açtım ve sd üzerinde bir baskı başlattım, gücü birkaç dakika sonra kestim ve işte!
> 
> conf_advc'de.
> 
> # define POWER_LOSS_RECOVERY 
> ETKİN (POWER_LOSS_RECOVERY) #if 
> # define PLR_ENABLED_DEFAULT yanlış 
> BACKUP_POWER_SUPPLY #define 
> // # POWER_LOSS_RECOVER_ZHOME tanımlamak 
> # define POWER_LOSS_ZRAISE 10 
> yerine pwrdet port # define POWER_LOSS_PIN P1_25 // i kullanılan e1det noktası 
> #define POWER_LOSS_STATE YÜKSEK 
> # define POWER_LOSS_PULL 
> POWER_LOSS_PURGE_LEN 20 #define 
> #define POWER_LOSS_RETRACT_LEN 10

Dostum problem marlin modunda değil zaten, grafik modunda çalıştırmak. Onu da ben yaptım zaten. Bu arada ben de Türküm 😄

berrenger, 2020-11-19:
> somehow i managed to get it worked.
>
> hardware: skr 1.4 t, relay and mini ups 24V, tft35 v3.0
> software: marlin bugfix 2.0.7
>
> all marlin configs are done according to instructions, i tried several [...]

Dostum problem marlin modunda değil zaten, grafik modunda çalıştırmak. Onu da ben yaptım zaten. Bu arada ben de Türküm 😄

ilkersismanoglu, 2020-11-19:
> Dostum problem marlin modunda değil zaten, grafik modunda çalıştırmak. Onu da ben yaptım zaten. Bu arada ben de Türküm 😄

tamam bu işlemden sonra grafik modunda da çalışmaya başladı zaten,

berrenger, 2020-11-19:
> tamam bu işlemden sonra grafik modunda da çalışmaya başladı zaten,

Peki elektrik kesintisinden sonra ve z yükselmesi işlemi bittiğinde z steplerinde çok küçük bir hareket oluyor mu? Ben de olmuştu. Elini z vidasına koyup kontrol eder misin?

ilkersismanoglu, 2020-11-20:
> Peki elektrik kesintisinden sonra ve z yükselmesi işlemi bittiğinde z steplerinde çok küçük bir hareket oluyor mu? Ben de olmuştu. Elini z vidasına koyup kontrol eder misin?

Bugün dediğinizi kontrol ettim herhangi bşr oynama yok..

Ancak başka bir sorunla karşılaştım dün deneme şansım olmamıştı.

Touch modda resume dediğimde baskı baştan başlıyor kaldığı yerden devam etmiyor, marlin modda hiç bir sıkıntı yok

Sanırım bu tft ekran touch modunda saçmalıyor

ilkersismanoglu, 2020-11-20:
> somehow i managed to get it worked.
>
> hardware: skr 1.4 t, relay and mini ups 24V, tft35 v3.0
> software: marlin bugfix 2.0.7
>
> all marlin configs are done according to instructions, i tried several [...]

New problem,

There’s something wrong with this tft35 v3.0

When tft in marlin mode, everything is perfect.  on power failure pausing print, raising z, sending kill code.. when power back on, click resume and printing continues perfectly from the last coordinates.

When tft in touch mode and power back on, 
it prompts me to resume. i choose ok, BUT printing starts from the beginning.

rakeshbarman, 2020-12-11:
Same problem here. working in marlin emulation mode but NOT in tft touch mode.
Srk-mini-e3-v2.0 + blTouchV2.1 + tft-e3-v3.0 + ups-24v-v1.0
Tested in marlin2.7.0.2 and now tested in bugfix-2.0.x (latest as of 8th Dec'20) with configuration files for ender-3
Interestingly, the latest bugfix-2.0.x has two new #define statement to choose from.
//#define POWER_LOSS_PULLUP         //#define POWER_LOSS_PULLDOWN
Just as a Test and taking a hint from https://reprap.org/forum/read.php?415,872921  that our sensor may need to be set up as pulldown. 
My present set of statements are
#define POWER_LOSS_RECOVERY
  #if ENABLED(POWER_LOSS_RECOVERY)
    #define PLR_ENABLED_DEFAULT   true
    #define BACKUP_POWER_SUPPLY 
    #define POWER_LOSS_RECOVER_ZHOME
    #define POWER_LOSS_ZRAISE       15 
    #define POWER_LOSS_PIN         PC12    // PWR_DET pin in Srk-mini-e3-v2.0
    #define POWER_LOSS_STATE     HIGH
    //#define POWER_LOSS_PULLUP
    #define POWER_LOSS_PULLDOWN
    #define POWER_LOSS_PURGE_LEN   20
    #define POWER_LOSS_RETRACT_LEN 10

Everything else other then in tft touch mode is working as intended.

rakeshbarman, 2020-12-11:
Same problem here. working in marlin emulation mode but NOT in tft touch mode.
Srk-mini-e3-v2.0 + blTouchV2.1 + tft-e3-v3.0 + ups-24v-v1.0
Tested in marlin2.7.0.2 and now tested in bugfix-2.0.x (latest as of 8th Dec'20) with configuration files for ender-3
Interestingly, the latest bugfix-2.0.x has two new #define statement to choose from.
//#define POWER_LOSS_PULLUP //#define POWER_LOSS_PULLDOWN
Just as a Test and taking a hint from https://reprap.org/forum/read.php?415,872921 that our sensor may need to be set up as pulldown.
My present set of statements are
#define POWER_LOSS_RECOVERY
#if ENABLED(POWER_LOSS_RECOVERY)
#define PLR_ENABLED_DEFAULT true
#define BACKUP_POWER_SUPPLY
#define POWER_LOSS_RECOVER_ZHOME
#define POWER_LOSS_ZRAISE 15
#define POWER_LOSS_PIN PC12 // PWR_DET pin in Srk-mini-e3-v2.0
#define POWER_LOSS_STATE HIGH
//#define POWER_LOSS_PULLUP
#define POWER_LOSS_PULLDOWN
#define POWER_LOSS_PURGE_LEN 20
#define POWER_LOSS_RETRACT_LEN 10

Everything else other then in tft touch mode is working as intended. So, still looking for a conclusion....

HTTYD9, 2020-12-11:
#define POWER_LOSS_PULLDOWN >> by changing it will fix , power loss recovery in tft mode ?

ilkersismanoglu, 2020-12-11:
The problem is solved for me..
I changed sd card configuration from lcd to onboard in marlin conf.

Now i can print any file from onboard sd via tft touch mode and everything works. Ups, relay, etc.

rakeshbarman, 2020-12-11:
thanks for the heads up. I will comment after understanding and trying it out.

rakeshbarman, 2020-12-11:
> #define POWER_LOSS_PULLDOWN >> by changing it will fix , power loss recovery in tft mode ?

No, that's just the chronology of my own tryouts. I think we should pay heed to what 'ilkersismanoglu' is trying to say about "changing sd card configuration from lcd to onboard in marlin conf." I will try it out first thing tomorrow and post results.

atharvbadkas, 2020-12-12:
> The problem is solved for me..
> I changed sd card configuration from lcd to onboard in marlin conf.
> 
> Now i can print any file from onboard sd via tft touch mode and everything works. Ups, relay, etc.

Hi @ilkersismanoglu ,can u send me your Marlin folder?
I'm using original LCD from Ender 3 Pro but Still not working for me, in combination with SKR MINI E3 V2 and BTT UPS 24V.

rakeshbarman, 2020-12-12:
> 
> 
> The problem is solved for me..
> I changed sd card configuration from lcd to onboard in marlin conf.
> 
> Now i can print any file from onboard sd via tft touch mode and everything works. Ups, relay, etc.
I checked my configuration_adv and found that it is already set as #define SDCARD_CONNECTION ONBOARD
Therefore didn't change anything. Back to square one, No power_loss_recovery from Touch Screen Mode,
So, still looking for a conclusion....

rakeshbarman, 2020-12-14:
> Same problem here. working in marlin emulation mode but NOT in tft touch mode.
> Srk-mini-e3-v2.0 + blTouchV2.1 + tft-e3-v3.0 + ups-24v-v1.0
> Tested in marlin2.7.0.2 and now tested in bugfix-2.0.x [...]

I made the following changes in the TouchScreenFirmware's config,ini and IT IS WORKING !
1. From https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware download the code.
2. Inside the code we will find a folder named "Copy to SD Card root directory to update"
3. Copy (1)   BIGTREE_TFT35_V3.0_E3.26.x.bin   (2) config.ini   in a FAT32 formatted miniSD card.
4. Open the folder "THEME_Unified Menu Material theme" and now copy the folder (3)  "TFT35" to the miniSD card.
5, Edit the file config.ini as below
   i)  find the line start_gcode:G28 XY R20\n  and change it to start_gcode:M75\nG28 XY R20\n
   ii) find the line end_gcode:M104 S0\nM140 S0\nM18\n and change it to end_gcode:M77\nM104 S0\nM140 S0\nM18\n
   iii)find the line cancel_gcode:M104 S0\nM140 S0\nG28 XY R10\n and change it to M77\nM104 S0\nM140 S0\nG28 XY R10\n
6. Save the file and insert the card in the TFT35 card reader and let it update.
   
   NB:  marlin command M75 = Start Print Job Timer and M77 = Stop Print Job Timer
          I think this inclusion was necessary as time keeper for our purpose

Everything is working as it should.
One thing to remember though is that we will not be prompted to resume print, After power fails during a print and thereupon when power resumes, wait for the machine to boot and just press the encoder button once. Printing start from where it left after Z_axis homing.
Kindly try and share your experience with this test.

----------------------------------------------------------------------------------------------------------------------------
In a different note, not related to TFTtouchscreen, I would also like to share that I made the following changes in marlin fw

Marlin/src/feature/powerloss.h

	  #define DEBUG_POWER_LOSS_RECOVERY // uncommented.
	  #define SAVE_EACH_CMD_MODE // uncommented.
	  //#define SAVE_INFO_INTERVAL_MS 0
---------------------------------------------------------------------

atharvbadkas, 2020-12-18:
Hi...I contacted BTT And they sent me files claiming they have tested it on BTT SKR MINI E3V2 AND BTT UPS 24V for Ender 3..but I'm still facing same issue with their firmware also....can u test it on your machine...and let me know??.... thanks...link https://drive.google.com/folderview?id=1lLJ8wB38pCyJNlhxEXdSRXJNUCHG2Eis

discip, 2021-02-25:
@HTTYD9
Some time ago you posted:
> I have same problem ,
> in tft mode UPS is not working (power loss recovery).
> [only relay works good , but relay with UPS not working]
> 
> BTT
> skr pro v1.1
> tft35 v3.0
> UPS 24V v1.0
> relay v1.2

Would you please be so kind, to tell me, how you achieved it, to make the ***'relay'*** work with the ***'skr pro'***?
Thank you in advance

DaStivi, 2021-02-25:
its working... UPS needs to be connected to SKR Board... setup up in firmware...
but also you need to enable "normal powerloss recovery" in TFT config.ini and also set the z raise height correctly there...
automatic startcode M75 and it should be fine then...

the "BTT UPS" setting in TFT doesn't do anything (yet?!)

emguepo, 2021-07-31:
> 
> 
> > Same problem here. working in marlin emulation mode but NOT in tft touch mode.
> > Srk-mini-e3-v2.0 + blTouchV2.1 + tft-e3-v3.0 + ups-24v-v1.0
> > Tested in marlin2.7.0.2 and now tested in bugfix-2.0.x (latest as of 8th Dec'20) with configuration files for ender-3
> > Interestingly, the latest bugfix-2.0.x has two new #define statement to choose from.
> > //#define POWER_LOSS_PULLUP //#define POWER_LOSS_PULLDOWN
> > Just as a Test and taking a hint from https://reprap.org/forum/read.php?415,872921 that our sensor may need to be set up as pulldown.
> > My present set of statements are
> > #define POWER_LOSS_RECOVERY
> > #if ENABLED(POWER_LOSS_RECOVERY)
> > #define PLR_ENABLED_DEFAULT true
> > #define BACKUP_POWER_SUPPLY
> > //#define POWER_LOSS_RECOVER_ZHOME
> > #define POWER_LOSS_ZRAISE 15
> > #define POWER_LOSS_PIN PC12 // PWR_DET pin in Srk-mini-e3-v2.0
> > #define POWER_LOSS_STATE HIGH
> > //#define POWER_LOSS_PULLUP
> > #define POWER_LOSS_PULLDOWN
> > #define POWER_LOSS_PURGE_LEN 20
> > #define POWER_LOSS_RETRACT_LEN 10
> > Everything else other then in tft touch mode is working as intended. So, still looking for a conclusion....
> 
> I made the following changes in the TouchScreenFirmware's config,ini and IT IS WORKING !
> 
>     1. From https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware download the code.
> 
>     2. Inside the code we will find a folder named "Copy to SD Card root directory to update"
> 
>     3. Copy (1)   BIGTREE_TFT35_V3.0_E3.26.x.bin   (2) config.ini   in a FAT32 formatted miniSD card.
> 
>     4. Open the folder "THEME_Unified Menu Material theme" and now copy the folder (3)  "TFT35" to the miniSD card.
>        5, Edit the file config.ini as below
>        i)  find the line start_gcode:G28 XY R20\n  and change it to start_gcode:M75\nG28 XY R20\n
>        ii) find the line end_gcode:M104 S0\nM140 S0\nM18\n and change it to end_gcode:M77\nM104 S0\nM140 S0\nM18\n
>        iii)find the line cancel_gcode:M104 S0\nM140 S0\nG28 XY R10\n and change it to M77\nM104 S0\nM140 S0\nG28 XY R10\n
> 
>     5. Save the file and insert the card in the TFT35 card reader and let it update.
>        NB:  marlin command M75 = Start Print Job Timer and M77 = Stop Print Job Timer
>        I think this inclusion was necessary as time keeper for our purpose
> 
> 
> Everything is working as it should.
> One thing to remember though is that we will not be prompted to resume print, After power fails during a print and thereupon when power resumes, wait for the machine to boot and just press the encoder button once. Printing start from where it left after Z_axis homing.
> Kindly try and share your experience with this test.
> 
> In a different note, not related to TFTtouchscreen, I would also like to share that I made the following changes in marlin fw
> 
> Marlin/src/feature/powerloss.h
> 
> ```
>   //#define DEBUG_POWER_LOSS_RECOVERY
>   #define SAVE_EACH_CMD_MODE // uncommented.
>   //#define SAVE_INFO_INTERVAL_MS 0
> ```

me ha funcionado. its works!