bigtreetech/BIGTREETECH-MINI-UPS-V2.0 #11: Marlin-2.0.7.2-SKR-mini-E3-V2.0 does not work with BTT UPS 24V V1.0

bartik, 2020-11-08:
Wiring:
The UPS is connected to PWR-DET (on the SKR-mini-E3-V2.0 board). The same is marked as PT-DET in the PIN description file. 

Configuration:
The Configuration_adv.h is set as in the documentation herein. The POWER_LOSS pin is taken from the source in btt git. It is the PC12 (value 44) configured in Configuration_adv.h (#define POWER_LOSS_PIN PC12). 

Reproduce:
When the power outage detection is turned on, the printer immediately after starting a print detects a power failure and doesnt start to print. It is only possible to turn the printer off.

Search:
Maybe this would help ? https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/144

My configuration file:
[Configuration_adv.h.txt](https://github.com/bigtreetech/BIGTREETECH-MINI-UPS-V2.0/files/5506525/Configuration_adv.h.txt)

atharvbadkas, 2020-11-16:
I'm Having same issue...

Jedrusia, 2020-11-27:
I have a identical problem and i spend many time with this problem. My problem exist with TFT35 with numerical LCD emulation mode and when i will print from SD Card plugged for SD slot on mainboard SKR MINI E3 V2.
Now i resolved my problem with printing in graphic mode of TFT and my file with gcode is loading from SD slot from TFT. This is working. If i plug the sd card with my gcode to SD slot on mainboard with graphical mode problem exist too.

atharvbadkas, 2020-12-12:
Hi @Jedrusia , which alternative supplier you were talking about??

Jedrusia, 2020-12-12:
MKS ?

atharvbadkas, 2020-12-18:
Hi...I contacted BTT And they sent me files claiming they have tested it on BTT SKR MINI E3V2 AND BTT UPS 24V for Ender 3..but I'm still facing same issue with their firmware also....can u test it on your machine...and let me know??.... thanks...link https://drive.google.com/folderview?id=1lLJ8wB38pCyJNlhxEXdSRXJNUCHG2Eis

Jedrusia, 2020-12-18:
I will test it for future, but now i work hardly and i not a  free time. sorry

killrob, 2020-12-30:
probem here too,
with mini E3 V2.0 and TFT35 in touch mode, i connected the UPS to "pwr-det" on E3 and modified the firmware as BTT says but on power loss Z don't raise, the restart print works well.
Regards.

atharvbadkas, 2020-12-30:
[image]

Check this out...
Is this useful??

Jedrusia, 2020-12-30:
> Hi...I contacted BTT And they sent me files claiming they have tested it on BTT SKR MINI E3V2 AND BTT UPS 24V for Ender 3..but I'm still facing same issue with their firmware also....can u test it [...]

This Yours firmware not workling for me

killrob, 2020-12-30:
> 
> 
> [image]
> 
> Check this out...
> Is this useful??

i will try now commenting the lines you say and let you know quite immediately, just the time to preheat

killrob, 2020-12-30:
Ok i tried the mods @atharvbadkas told but still don't work, Z is not rising

Jedrusia, 2021-05-05:
This solution not working on my printer.

ivanroberto, 2021-05-27:
Hey guys, in my case I have SKR PRO V1.1 and it is behaving the same way, I alread tried E1, E2 and Z_MIN connectors (PE10, PG5 and PG8). 
Now I tried trigger with a direct wire from these same pins above without use UPS, just to be sure if the problem was my UPS, but NO, there is no Z raise....

I supect that issue is a bug on marlin firmware. I tryed 2.0.8, 2.0.8.1, bugfix, but no one give me the raise on Z.

```
#define POWER_LOSS_RECOVERY
  #if ENABLED(POWER_LOSS_RECOVERY)
    #define PLR_ENABLED_DEFAULT   false    // Power Loss Recovery enabled by default. (Set with 'M413 Sn' & M500)
    #define BACKUP_POWER_SUPPLY           // Backup power / UPS to move the steppers on power loss
    #define POWER_LOSS_ZRAISE     10     // (mm) Z axis raise on resume (on power loss with UPS)
    #define POWER_LOSS_PIN        PG5 // Pin to detect power loss. Set to -1 to disable default pin on boards without module.
    #define POWER_LOSS_STATE      HIGH // State of pin indicating power loss
    #define POWER_LOSS_PULLUP         // Set pullup / pulldown as appropriate for your sensor
    //#define POWER_LOSS_PULLDOWN
    #define POWER_LOSS_PURGE_LEN   20 // (mm) Length of filament to purge on resume
    #define POWER_LOSS_RETRACT_LEN 10 // (mm) Length of filament to retract on fail. Requires backup power.

    #define POWER_LOSS_MIN_Z_CHANGE 0.05 // (mm) Minimum Z change before saving power-loss data

    //#define POWER_LOSS_RECOVER_ZHOME
    #if ENABLED(POWER_LOSS_RECOVER_ZHOME)
      //#define POWER_LOSS_ZHOME_POS { 0, 0 } // Safe XY position to home Z while avoiding objects on the bed
    #endif
  #endif
```
So, on the same boat as you....

Jedrusia, 2021-05-28:
I don't have any working solution now.

ivanroberto, 2021-05-28:
Still in time, i forgot to mention that i also included the M75 command in the start script and in the M77 stop/cancel script. Even so, there is no lifting of the Z axis.

Marcus9812, 2021-06-06:
For some reason, my UPS raises the z-axis and performs the power loss function, but when I try to continue printing, it goes home, moves to my part location, and stops moving (I use BTT GTR V1.0).

Hassan647539, 2021-11-12:
I have the same problem with SKR1.3 with Marlin-2.0.7.2 and BTT UPS 12 V