bigtreetech/BIGTREETECH-MINI-UPS-V2.0 #14: Not working on Marlin 2.7.1.2 SKR1.4

Ejay-3D, 2021-02-09:
Hi, 
I installed this module on a BIqu B1 that have a skr 1.4 board. 
The wiring is correctly done and the frimware instruction followed correctly as well . 
While printing, if I remove the power cable : 
- The Z Does not Raise 
- The machine shutdown after 1 seconds 
- After resuming the print restart much more earlier than when it stopped, with a beautiful blob on the print due to the fact Z Didn't raise. 

Conclusion With or without the module I don't see much difference, the only difference is that the print continues a bit more time ... than without ... 

Any advices?

Ejay-3D, 2021-03-02:
Is anyone from BTT looking a those Git issues ? 

The clear thing is that on a SKR1.4 using MArling 2.7.1.2 , the resume gcode status is not saved therefore it's not possible to resume a print when printing from mother board. 

On the Screeen ( since the screen is not aware of the UPS) It doesn't do what it should before shutdown then the module is unusefull as well .

Cabinfourus, 2021-03-07:
Try adding M75 to your start gcode manually through your software. Also add M77 to end gcode.

Ejay-3D, 2021-07-10:
Still the Same issue With The Biqu BX SKR 2.0 . 
The Z Doesn't Raise on power failure , but goes down of 10 after resuming ... 
Sad to see that Btt is not looking at the issue opened here ...