bigtreetech/BIGTREETECH-MINI-UPS-V2.0 #15: Print Halted immediately after starting

Joe-githubby, 2021-02-13:
I have not been able to get the UPS working with SKR Mini E3 V2. 

When I start a print, it immediately says print halted. I have correctly configured the file as per the manual and other suggestions online but it doesnt seem to remedy the situation. 

It is worth noting that the electricity is not turned off. It just detects power outage for no reason. It does not allow to print at all. 

**Configuration File**
[Config.txt](https://github.com/bigtreetech/BIGTREETECH-MINI-UPS-V2.0/files/5976287/Config.txt)

**Steps to reproduce**
Print file from on board SD

**Expected behavior**
Print file as selected

**Actual behavior**
Power outage detected, print halted.

cookins, 2021-02-17:
maybe you got some pins, cable - both on motherboard or ups - wrongly soldered, as it bought? try to change:
to
`#define POWER_LOSS_STATE     LOW`

Joe-githubby, 2021-02-27:
this did not work. It stopped printing immediately after selecting a file to print.

cookins, 2021-02-27:
same as before, then?

sinole1987, 2021-04-05:
Hi
I noticed the problem too, the power panic function is way too sensitive, it detects smallest power fluctuations, if your power supply is not the best, it might not have the smoothest power delivery, or AC power might go out sometimes in such small interval of time that it doesn't effect any device in your home. my assumption is that this module detects all of that and just stops a print that can actually go on without this module. mine does not stop prints immediately but it will trigger somewhere mid print for no reason. so I kept it connected to power supply so it can save prints from power flickers, but I disconnected power outage functionality . not many power outages in my area. capacitor's are great, power panic module not so much.
ultimately you can buy prusa power panic plug and in combination with this ups they will work great.