bigtreetech/BIGTREETECH-MINI-UPS-V2.0 #18: UPS Stops Printer from booting.

sinole1987, 2021-04-02:
whenever printer is off for a long time and capacitors become empty, they draw so much current that printer doesn't boot when I turn it on. I have to  turn off and wait for screen to stop flickering and then turn it on again. even soft reset doesn't work. I bought several of these modules they all have this problem. I think they need to be revised to draw less current when they are empty.

makmillion, 2021-07-26:
I, too, am facing this issue. I'm doing what you describe, but there should be some included way to avoid this.

sinole1987, 2021-07-26:
well use longer wire than the one is going to your printer. significantly longer wire or you van use lighter gauge as well which is a little dangerous. I am just using a very long 16 awg wire from psu to ups and hide the wire to my designed enclosure. that increases the resistance enough to dampen the draw.

uskebasy, 2021-09-21:
I have the same issue, how we can solve this?

Voodoobrew101, 2021-09-29:
I think this is because my 24v 600w power supply is not powerful enough to charge the capacitors which have a huge initial amp draw. 

After some searching it seems a decent power supply should have this built in over current protection that should prevent this, 
or maybe because it is there this happens? It seems like it should be built in to this product but maybe now a separate overcurrent protection circuit is needed? like this https://www.youtube.com/watch?v=e

Does that sound like what's going on here?

uskebasy, 2021-09-29:
> I think this is because my 24v 600w power supply is not powerful enough to charge the capacitors which have a huge initial amp draw.

The problem is not the power supply in your case, 600W is very much power. The problem is that the UPS adsorb too much current at the beginning and the overcurrent protection of the PSU opens the circuit.

> After some searching it seems a decent power supply should have this built in over current protection that should prevent this, or maybe because it is there this happens? It seems like it should be built in to this product but maybe now a separate overcurrent protection circuit is needed? like these https://www.homemade-circuits.com/universal-high-watt-led-current-limiter/ or this https://www.youtube.com/watch?v=e or maybe just a big resistor that will also lower the voltage though (right??). Does that sound like what's going on here?

We need a solution that reduces the initial current, not the voltage. I don't know if the link you provided can be used to limit the current in our case.

> That is exactly what is happening. "overcurrent protection circuit" Sounds like a fuse to me.

I think that it is a relay, not a fuse, that opens the circuit to cut the current (I hear the "click" noise of the relay that repetitively open the circuit.)

> you need something to reduce current not to cut it. what I did was a longer cable to the capacitor, and it is working great. It is not dangerous because high current is momentary and it drops immediately before cable even get slightly warm.

Your solution is interesting, at least it solves the problem. The long cable is like a resistance in the circuit, but can we found a "better" solution? A proper resistance or a circuit that reduces the current? I would like to avoid this very long cable under the printer. Anyway, how long is the cable? I will try your solution, until we find another one, because the UPS is useless right now, I don't know how all the other people can use it.

sinole1987, 2021-09-29:
I increased the cable length around 4-5 feet LOL , and pushed it into a box. I don't like it either but that was the only thing I found working. the idea was to make UPS cable slightly longer than cable that goes to motherboard. you can also reduce the length of the cable to your motherboard. because they are running parallel and electricity takes the path of lower resistance. BTT in some videos actually suggest that you run the UPS in series , which is very messy. meaning that you run the cable from motherboard inlet to your UPS, instead of running it from PSU to UPS directly.

uskebasy, 2021-09-29:
Do you mean the total length of positive + negative cables or the length of the single cable?. So it is about 1.5 meters, I can try.
The cable of the motherboard I guess that is less than 0.4 meters.
As regards the connection in series I don't get the point, can you share these videos?

sinole1987, 2021-09-29:
I would reduce both as much as I can. I didnt really calculate, I just know it would work. unfortunately you goto just run try and error.  I wouldn't recommend trying with cheaper cable. because again when you use better cable it has lower resistance.  but you can try cutting "+" cable first and see if it helps. but before that try the series method and see if it helps. the problem is current draw is different from UPS to UPS, i had to purchase a few of these and test and choose the one with lower draw.

uskebasy, 2021-09-29:
Did you measure the current with a multimeter? How much current draws?

sinole1987, 2021-09-29:
I don't think you can truly measure the initial draw, it is ramps down really fast. you might need a better equipment than multimeter.

uskebasy, 2021-09-29:
I have the original PSU that is 250W, it is another user that has the 600W PSU. Do you have problems also with the 360 watts PSU?

sinole1987, 2021-09-29:
yes, I used to, before I used longer cables

Sergeykurm, 2022-12-05:
I think that BTT is doing this specifically for the development of the UPS line. they had to add a relay to the UPS.
For now, there is one safe option. Assemble the time delay relay according to the scheme (3-5 seconds). [Or buy on aliexpress](https://fr.aliexpress.com/item/1005004979955717.html?spm=a2g0o.productlist.0.0.78e2279e1KZzAY&algo_pvid=bf181d4d-b3bd-4b80-8609-5f54a2c1abe0&algo_exp_id=bf181d4d-b3bd-4b80-8609-5f54a2c1abe0-48&pdp_ext_f=%7B"sku_id"%3A"12000031224702260"%7D&pdp_npi=2%40dis%21EUR%215.64%215.13%21%21%21%21%21%402100bb4916702407446251441e1e8f%2112000031224702260%21sea&curPageLogUid=2Hg7MCF4X6wa).

uskebasy, 2022-12-05:
> When you turn on the printer, the supercapacitors are charged. After 3-5 seconds, the printer turns on.

I think that this will not solve the problem. The problem is that the PSU overcurrent protection prevents the Supercapacitors to absorb the current they want, so. Retarding the power on of the printer is not a solution to this problem.

Sergeykurm, 2022-12-05:
> I think that this will not solve the problem. The problem is that the PSU overcurrent protection prevents the Supercapacitors to absorb the current they want, so. Retarding the power on of the [...]

#
[image]
 I think this will solve the problem!

uskebasy, 2022-12-05:
Thank you for this image, can you post the source? I want to further investigate to better understand.
If this work, also a simple interrupter (manual) will do the work

Sergeykurm, 2022-12-05:
> Thank you for this image, can you post the source? I want to further investigate to better understand. If this work, also a simple interrupter (manual) will do the work

I myself am trying to solve this issue of correct inclusion. I came up with a solution. Drawn in a quick hand for understanding. Or put the second button and turn it on and off with two buttons.

uskebasy, 2022-12-05:
Have you tried this solution? Does it work?

Sergeykurm, 2022-12-05:
> Have you tried this solution? Does it work?

The button works. Relay ordered. I am waiting.

G-many2k, 2022-12-06:
have a look at this clip it could help i orderd a few waithing for the then will wire it up 
https://www.youtube.com/watch?v=-HDpkn8JT_0

Sergeykurm, 2022-12-06:
> 
> I don't think you are right, we are talking exactly of high starting current. The supercapacitor at the start consumes a current that is higher than the maximum load supported by the PSU. Indeed, I don't think your solution with the interrupter could work with all PSU.

The capacitor will draw as much current as it needs. If the power supply is 350W, the charge will be a little slower, if the power supply is 600W, the capacitors will charge faster. That's the difference. When the capacitor is charged, the current from the power supply and from the capacitors equalizes and is supplied to the MB after 3 seconds without drops.

uskebasy, 2022-12-06:
> The capacitor will draw as much current as it needs. If the power supply is 350W, the charge will be a little slower, if the power supply is 600W, the capacitors will charge faster. That's the [...]

Have you read all the posts in this discussion? The capacitors WOULD draw as much current as they need, but the PSU overcurrent protection prevents the capacitors to draw as much current, and it cuts the circuit with a fuse or relay. This keeps going on until the capacitors are charged.
So, I don't think that your solution with the interrupter can prevent this, you are only activating the motherboard after the capacitors are charged. But the PSU is still "soffering" and cutting the circuit because of the high starting current.

sinole1987, 2022-12-08:
solution is very simple V=IR, v always constant increase the R will reduce I. when over current protection triggers that means you have too much current, reduce it. I did that by increasing the length of the cable, not the best of solution but it still works.
That also means current draw is not much higher than PSU can handle. 
you have better solution let me know! maybe a resistor with low resistance somewhere on live wire that goes to PSU.

Sergeykurmelev, 2022-12-10:
[image]

Added an emergency stop button.

Everything is working!!!

sinole1987, 2022-12-11:
I do have a relay and that doesn't help anything, problem is mostly internal in power supply circuit. It simply doesnt allow the massive draw from fully empty capacitors . so when you initially turn on the printer PSU go to stall. you reset the psu, now capacitors are not as hungry so they don't draw like they did before. and printer works. what we need is a way to slow down that initial draw.

Sergeykurm, 2022-12-11:
Maybe it's all about the power supply. I have MW 350w. It works for me according to the scheme above.

uskebasy, 2022-12-11:
@Sergeykurmelev
> Maybe it's all about the power supply. I have MW 350w. It works for me according to the scheme above.

You have a PSU that can supply the current drawn by the caps. Your solution is useful only to charge the caps and then start the motherboard. This does not resolve the general problem of this UPS, as I am trying to explain to you.

@sinole1987 
> I did that by increasing the length of the cable, not the best of solution but it still works.

I did this when you advised me to do so, but this does not work for me.

> That also means current draw is not much higher than PSU can handle.

Maybe, in my case, the PSU cannot handle the current draw also if it is a little lower (it is the stock crap PSU of the TEVO Tarantula)

> you have better solution let me know! maybe a resistor with low resistance somewhere on live wire that goes to PSU.

I think that the solution proposed by @G-many2k [above]( https://github.com/bigtreetech/BIGTREETECH-MINI-UPS-V2.0/issues/18#issuecomment-1338964899) could work, but I can not try right now.

sinole1987, 2022-12-12:
> @Sergeykurmelev
>
>
> You have a PSU that can supply the current drawn by the caps. Your solution is useful only to charge the caps and then start the motherboard. This does not resolve the general [...]

yes, that works, I just watch that video. I didn't want to recommend a resistor, because it could be very dangerous. even long wire is dangerous to an extend, its a resistor at the end of the day, but I tested it and wires never even got hot to touch for me.

Sergeykurm, 2022-12-12:
So it is necessary to implement such a scheme.

[image]

A powerful resistor limits the current for consumption by the capacitors, after the capacitors are charged, the relay is activated, turning off the resistor so that it does not heat up. So much safer than installing a thermistor.

G-many2k, 2022-12-12:
> So it is necessary to implement such a scheme.
> 
> [image]
> 
> A powerful resistor limits the current for consumption by the capacitors, after the capacitors are charged, the relay is activated, turning off the resistor so that it does not heat up. So much safer than installing a thermistor

Please provide more information on the Resistor relay set up you have draw above .
have you tested your theory ?

uskebasy, 2022-12-12:
> A powerful resistor limits the current for consumption by the capacitors, after the capacitors are charged, the relay is activated, turning off the resistor so that it does not heat up. So much safer than installing a thermistor.

This should work, and you can test it very quickly. But the best solution is to have a thermistor in place of the resistor in your schema.

kalpeke, 2023-04-18:
I tested with a 3A NTC current limiting thermistor and I no longer get the clicks but sometimes during a save the nozzle fails to lift(but the print is still saved) - I think 3 amp is too small; I then tried a 5A thermistor and I get 2 to 3 clicks but the nozzle lifts as expected during a save. 
The other is issue is when you turn off the printer and it runs from the UPS at some point when the printer is draining the capacitors the LEDs start flickering as the voltage goes down untill the capacitors are completely drained then the LEDs turn off(Duet3d by the way) - the motors also make this grinding noise. I think to fix this issue a low voltage cutoff module can be installed but I don't have one so its not tested.

```
specs:
power supply Delta 150W 6.25A
3A thermistor SCK333 https://www.communica.co.za/products/sck15333msy
5A https://www.communica.co.za/products/sck20100msby?variant=20058344161353
```
when using the 5A thermistor the shutdown was more gracefull (LEDs did not flicker as much)
[image]
[image]

PS this thing should have never went to mass production how were this issues not pickedup during testing