bigtreetech/BIGTREETECH-MINI-UPS-V2.0 #20: Documentation out dated, Marlin should be #define POWER_LOSS_PULLUP or #define POWER_LOSS_PULLDOWN  ?

badbod, 2021-09-03:
While I can try both and see what works, I though it might also be an idea to have the documentation tell me.

should it be 
  #define POWER_LOSS_PULL pulldown
OR
  #define POWER_LOSS_PULL pullup

For Marlin 2.0.x  which is correct ? 
  #define POWER_LOSS_PULLUP
OR
  #define POWER_LOSS_PULLDOWN

Is it the same for UPS 24V V1.0 ?  (for which I can find no instructions at all)

rebel1985, 2022-03-27:
i have the same question

Windbane, 2022-06-19:
[image]
 It's PULLUP, I took this image from BTT's SKR3 manual.

Windbane, 2022-12-08:
> Data in the manual doesn't work

#define POWER_LOSS_PULLUP

Escrich, 2022-12-12:
[image]
I think that everything could work, my feelings are that the sum of Power supply and UPS, are not enough to complete the power down cicle, Raise 10 millimeters the nozzle it's a big energy spend, I try to raise it only two, of course you must have the SD active, as in the attached image, then in those conditions I got sometimes a file on the SD card, but it's empty and unreadable, my though is that this is happening because of a lack of power, Marlin must reaction in a different way than now does, just in the moment that the power down signal appears, must disconnect heating, every heating and every fan, must stop any motion, except the one to raise at least a couple of millimeters the Z axis, in my opinion, after lot of tries and changes, the system works, but need some more extra power, my settings are for an Octopus V1.1 Card made from BTT, and the 24 Volts UPS from BTT too, I hope in a near future Marlin will improve the related routines, there are some people with those UPSs working now, then and because of that, I think is difficult that Marlin willmake changes for that